DFiantHDL/DFHDL #137: prev/prev(1) is not correctly assigned to VARs

MartinC45, 2024-05-13:
## Description
Assigning .prev / .prev(1) to an OUT gives the expected result, but assigning it to a VAR doesn't.  In the below example, the Verilog code has a being assigned to c directly instead of a_reg1. For higher orders, it works as expected, as shown by e.

## Example
```scala
import dfhdl.*

class FirstDF() extends DFDesign:
    val a = Bit <> IN init 0
    val c = Bit <> VAR 
    val e = Bit <> VAR
    c := a.prev(1)
    e := a.prev(2)
    
    val d = Bit <> OUT init 0
    d := a.prev

@main def main = 
    FirstDF().compile
```
## Output
```verilog
module FirstDF(
  input wire logic clk,
  input wire logic rst,
  input wire logic a,
  output logic     d
);
  logic c;
  logic e;
  logic a_reg1;
  logic a_reg2;
  always @(*)
  begin
    e        = a_reg2;
    d        = a_reg1;
  end
  always @(posedge clk)
  begin
    if (rst == 1'b1) begin
      a_reg1 <= 1'b0;
      a_reg2 <= 1'b0;
      c      <= 1'b0;
    end
    else begin
      a_reg1 <= a;
      a_reg2 <= a_reg1;
      c      <= a;
    end
  end
endmodule
```

MartinC45, 2024-05-13:
Technically, it is correct, as c assumes the value of a one cycle later. But c is in the clocked process, which is not the case for OUT or for higher orders and a bit unexpected.

soronpo, 2024-05-14:
In dataflow you can only assume dependency and insertion order. There are no clocks in DF. Although this could suggest something unplanned is happening there due to the different behavior, but there is no descrepency with DF rules in this example alone.

soronpo, 2025-01-13:
@MartinC45 
DFHDL v0.9.0 gives you this:
```verilog
`default_nettype none
`timescale 1ns/1ps
`include "FirstDF_defs.svh"

module FirstDF(
  input  wire logic clk,
  input  wire logic rst,
  input  wire logic a,
  output      logic d
);
  `include "dfhdl_defs.svh"
  logic c;
  logic e;
  logic a_reg1;
  logic a_reg2;
  always_ff @(posedge clk)
  begin
    if (rst == 1'b1) begin
      d      <= 1'b0;
      c      <= 1'b0;
      a_reg1 <= 1'b0;
      a_reg2 <= 1'b0;
    end
    else begin
      a_reg1 <= a;
      a_reg2 <= a_reg1;
      c      <= a;
      d      <= a;
    end
  end
  assign e = a_reg2;
endmodule
```
You check it out in this scastie:
https://scastie.scala-lang.org/egYDpMY9TtaOoPrld1K2FQ

Is this more to your liking?

soronpo, 2025-01-17:
This issue is not a bug, even before the change, but now I hope the generation makes it a little clearer.

closed 2025-01-17 by soronpo