DFiantHDL/DFHDL #147: Clock and reset signals of submodules declared but not assigned in RTDesign

MartinC45, 2024-05-18:
## Example
```scala
val clkCfg1 = ClkCfg(ClkCfg.Edge.Rising)
val rstCfg1 = RstCfg(RstCfg.Mode.Sync, RstCfg.Active.High)
val rtcfg = RTDomainCfg(clkCfg1, rstCfg1)

class INV() extends RTDesign:
    val a = Bit <> IN
    val c = Bit <> VAR.REG init 0
    val b = Bit <> OUT
    c.din := a
    b := !a && c
    
class ClockRstConnection() extends RTDesign(rtcfg):
    val a = Bit <> IN
    val b = Bit <> OUT
    
    val inv0 = new INV()
    
    inv0.a <> a
    inv0.b <> b

@main def main = 
   ClockRstConnection().compile
```

## Output
```verilog
`default_nettype none
`timescale 1ns/1ps
`include "ClockRstConnection_defs.sv"

module ClockRstConnection(
  input wire logic clk,
  input wire logic rst,
  input wire logic a,
  output logic     b
);
  logic inv0_clk; // declared but never assigned
  logic inv0_rst;
  logic inv0_a;
  logic inv0_b;
  INV inv0(
    .clk /*<--*/ (inv0_clk),
    .rst /*<--*/ (inv0_rst),
    .a   /*<--*/ (inv0_a),
    .b   /*-->*/ (inv0_b)
  );
  assign inv0_a = a;
  assign b      = inv0_b;
endmodule
```

soronpo, 2024-06-11:
@MartinC45 apologies for the delay in fixing this.

closed 2024-06-11 by soronpo

MartinC45, 2024-06-11:
No problem at all, thanks for the fix. With this, I could get a first slightly larger [design](https://github.com/MartinC45/LZW_DFHDL) involving multiple modules working, without any VHDL edits required to run simulation.

commit f8bc367a (2024-06-11): fix #147 (add regression test)