DFiantHDL/DFHDL #148: Clock and reset missing in generated files when top level is combinational

MartinC45, 2024-05-18:
## Descriptions
Clock and reset are missing in the generated code, when the top level class has both no REG and no explicit RTDomainCfg (that is, only compiler config is given). Basically, when the top level is combinational.

## Example
```scala
class INV() extends RTDesign():
    val a = Bit <> IN
    val c = Bit <> VAR.REG init 0
    val b = Bit <> OUT
    c.din := a
    b := !a && c
    
class NoCfgClockRst() extends RTDesign(): // if RTDesign(cfg) it works fine
    val a = Bit <> IN
    val b = Bit <> OUT // with OUT.REG it works
    
    val inv0 = new INV()
    
    inv0.a <> a
    inv0.b <> b

given options.CompilerOptions.DefaultClkCfg = ClkCfg(ClkCfg.Edge.Rising)
given options.CompilerOptions.DefaultRstCfg = RstCfg(RstCfg.Mode.Sync, RstCfg.Active.Low)

@main def main = 
   NoCfgClockRst().compile
```

## Output
```verilog
`default_nettype none
`timescale 1ns/1ps
`include "NoCfgClockRst_defs.sv"

module INV(
  input wire logic a,
  output logic     b
);
  logic c;
  logic c_din;
  always @(*)
  begin
    c_din = c;
    c_din = a;
    b     = (!a) & c;
  end
endmodule
```

```verilog
`default_nettype none
`timescale 1ns/1ps
`include "NoCfgClockRst_defs.sv"

module NoCfgClockRst(
  input wire logic a,
  output logic     b
);
  logic inv0_a;
  logic inv0_b;
  INV inv0(
    .a /*<--*/ (inv0_a),
    .b /*-->*/ (inv0_b)
  );
  assign inv0_a = a;
  assign b      = inv0_b;
endmodule
```

soronpo, 2024-05-19:
This is the intended behavior. As you figured out, by explicitly applying a clock configuration, the clk/rst signaling is added. If there is no explicit configuration and no clk is required the design is purely combinational without clk/rst signaling.

closed 2024-05-19 by soronpo

soronpo, 2024-07-06:
@MartinC45 are you available for chatting about clk/rst configuration in the coming days?